matlo/serialusb #5: Failed to start the proxy: initialization timeout expired

marcel-bootsma, 2019-02-12:
Hello Mathieu,

I am trying to setup your serial usb proxy but I keep getting a timeout error (see below).
I am running Linux Mint:
  System:    Host: DellXPSMint Kernel: 4.15.0-33-generic x86_64 (64 bit)
  Desktop: Cinnamon 3.6.7  Distro: Linux Mint 18.3 Sylvia

I use a PL2303TA USB to serial converter an a pro micro clone.
I tried with another device an promicro with the same result
I get the firmware in the pro micro OK using the gimx launcher.

What could be wrong?

marcel@DellXPSMint ~ $ sudo serialusb-capture.sh 
Select a USB to UART adapter:
0: /dev/ttyUSB0
0
Selected: /dev/ttyUSB0
Overwrite /home/marcel/capture.pcap? [y/n]
y
Available USB devices:
 0 VID 0x1d6b (Linux Foundation) PID 0x0003 (3.0 root hub) PATH 02
 1 VID 0x04f3 (Elan Microelectronics Corp.) PID 0x21d5 (unknown product) PATH 01:09
 2 VID 0x0a5c (Broadcom Corp.) PID 0x6410 (unknown product) PATH 01:04
 3 VID 0x06f7 (Wailly Technology Ltd) PID 0x0003 (USB->Din 4 Adaptor) PATH 01:02:03
 4 VID 0x046d (Logitech, Inc.) PID 0xc05b (M-U0004 810-001317 [B110 Optical USB Mouse]) PATH 01:02:01
 5 VID 0x05e3 (Genesys Logic, Inc.) PID 0x0608 (Hub) PATH 01:02
 6 VID 0x1bcf (Sunplus Innovation Technology Inc.) PID 0x2b95 (unknown product) PATH 01:0c
 7 VID 0x067b (Prolific Technology, Inc.) PID 0x2303 (PL2303 Serial Port) PATH 01:01
 8 VID 0x1d6b (Linux Foundation) PID 0x0002 (2.0 root hub) PATH 01
Select the USB device number: 3
Opened device: VID 0x06f7 PID 0x0003 PATH 01:02:03
configuration: 1
  interface: 0:0
    endpoint: IN INTERRUPT 1
proxy.c:727 proxy_start: Failed to start the proxy: initialization timeout expired!
Capture failed.
marcel@DellXPSMint ~ $

matlo, 2019-02-12:
Could be either incorrect wiring or the PL2303TA not supporting 500kbps (or with an error rate above 0).

It also seems you are not using latest version (0.7).

closed 2019-02-12 by matlo

marcel-bootsma, 2019-02-12:
Hello Mathieu,

Thank you for the quick response!
I did not find any other way to contact you therefore the issue.

I tried the latest version first without success and thought I try an older version also.
I will buy a different USB2Serial and Pro micro and test with them.

Is it possible to use a lower speed? I just use a 1.5Mb device.

Thank you again for helping.

matlo, 2019-02-12:
Changing the baudrate involves modifying both firmware and software. I'm too busy to help on this.

marcel-bootsma, 2019-02-12:
OK, sorry for the disturbance.
I will try other hardware and let you know when it works.

marcel-bootsma, 2019-02-12:
I found a FT232 on a Arduino Duemilanove.
I ran a simple Arduino sketch to make sure the serial connection works both ways.
Now your serialusb capture works!
I think the PL2303TA is just not good enough for this speed.

Thank you

matlo, 2019-02-12:
Happy hacking :)

lmirel, 2019-06-03:
I stumbled upon an issue generating the same behavior, but for different reasons:
the tool will timeout when ran by itself (no .sh script) if the parameter used for the serial adapter is provided via /dev/serial/by-id/.. for example

$ ls -la /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0
lrwxrwxrwx 1 root root 13 May 18 13:13 /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB3

so if the tool is run with /dev/serial/by-id/usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 instead of /dev/ttyUSB3 it will timeout.

$ ls -la /dev/ttyUSB3
crw-rw---- 1 root dialout 188, 3 May 18 16:55 /dev/ttyUSB3

so I used realpath before opening the device:

> int adapter_open(const char * port, ADAPTER_READ_CALLBACK fp_read, ADAPTER_WRITE_CALLBACK fp_write, ADAPTER_CLOSE_CALLBACK fp_close) 
{
  char tbuf[PATH_MAX];
  int serial = -1;
  if (realpath (port, tbuf))
    serial = gserial_open (tbuf, USART_BAUDRATE);

matlo, 2019-06-03:
gimxserial expect /dev/tty* or /dev/spi* devices. It probably should issue an error if the provided device does not meet this expectation.

See https://github.com/matlo/gimxserial/blob/master/src/posix/gserial.c#L184