matlo/serialusb #6: async.c:82 open failed with error: No such file or directory

blshkv, 2019-02-15:
Hi, any idea what's wrong?
```
lsusb
Bus 003 Device 036: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC

```
```
 sudo serialusb-capture.sh 
Available USB to UART adapters:
0: /dev/ttyUSB0
Select the USB to UART adapter number: 0
Selected: /dev/ttyUSB0
Available USB devices:
 0 VID 0x8087 (Intel Corp.) PID 0x0024 (Integrated Rate Matching Hub) PATH 04:01
 1 VID 0x1d6b (Linux Foundation) PID 0x0002 (2.0 root hub) PATH 04
 2 VID 0x04f2 (Chicony Electronics Co., Ltd) PID 0xb2ea (Integrated Camera [ThinkPad]) PATH 03:01:06
 3 VID 0x0a5c (Broadcom Corp.) PID 0x21e6 (BCM20702 Bluetooth 4.0 [ThinkPad]) PATH 03:01:04
 4 VID 0x0403 (Future Technology Devices International, Ltd) PID 0x6001 (FT232 Serial (UART) IC) PATH 03:01:02
 5 VID 0x8087 (Intel Corp.) PID 0x0024 (Integrated Rate Matching Hub) PATH 03:01
 6 VID 0x1d6b (Linux Foundation) PID 0x0002 (2.0 root hub) PATH 03
 7 VID 0x1d6b (Linux Foundation) PID 0x0003 (3.0 root hub) PATH 02
 8 VID 0x413c (Dell Computer Corp.) PID 0x301a (unknown product) PATH 01:01
 9 VID 0x1d6b (Linux Foundation) PID 0x0002 (2.0 root hub) PATH 01
Select the USB device number: 4
Opened device: VID 0x0403 PID 0x6001 PATH 03:01:02
configuration: 1
  interface: 0:0
    endpoint: IN BULK 1
      endpoint 1 won't be configured (not an INTERRUPT endpoint)
    endpoint: OUT BULK 2
      endpoint 2 won't be configured (not an INTERRUPT endpoint)
lib/gasync/src/common/linux/async.c:82 open failed with error: No such file or directory
Capture failed.
```

blshkv, 2019-02-15:
well, I guess that's because I do not have a atmega32u4 device in a chain. You might want to make an additional check to display a clear message

matlo, 2019-02-15:
Due to how UART works there is no way to tell if nothing is connected to the USB to UART adapter. The atmega32u4 is listed in the requirements.

closed 2019-02-15 by matlo