matlo/serialusb #7: Faild to load and init proxy

slmaxk, 2019-03-12:
Hello Matlo,

i tried to Build your Serial-USB, flash the ATMega. Install Your branch. But i have an Issue. Can you help me to fix that. 

 Sorry for my bad English and dumb Questions. 

thanks in advance 
Franjo / slmaxk

"
franjo@sernoevil:~/serialusb/sw$ sudo serialusb-capture.sh
Available USB to UART adapters:
0: /dev/ttyUSB0
Select the USB to UART adapter number: 0
Selected: /dev/ttyUSB0
Overwrite /home/franjo/serialusb/sw/capture.pcap? [y/n] y
Available USB devices:
 0 VID 0x8087 (Intel Corp.) PID 0x0020 (Integrated Rate Matching Hub) PATH 02:01
 1 VID 0x1d6b (Linux Foundation) PID 0x0002 (2.0 root hub) PATH 02
 2 VID 0x0a5c (Broadcom Corp.) PID 0x217f (BCM2045B (BDC-2.1)) PATH 01:01:04
 3 VID 0x147e (Upek) PID 0x2016 (Biometric Touchchip/Touchstrip Fingerprint Sensor) PATH 01:01:03
 4 VID 0x1430 (RedOctane) PID 0x0150 (wireless receiver for skylanders wii) PATH 01:01:02
 5 VID 0x0403 (Future Technology Devices International, Ltd) PID 0x6001 (FT232 Serial (UART) IC) PATH 01:01:01
 6 VID 0x8087 (Intel Corp.) PID 0x0020 (Integrated Rate Matching Hub) PATH 01:01
 7 VID 0x1d6b (Linux Foundation) PID 0x0002 (2.0 root hub) PATH 01
Select the USB device number: 4
Opened device: VID 0x1430 PID 0x0150 PATH 01:01:02
proxy.c:702 plugin_init: failed to load plugin
proxy.c:731 proxy_start: failed to init plugin
Capture failed.
franjo@sernoevil:~/serialusb/sw$

matlo, 2019-03-12:
Hello,
There is no plugin_init function in current source code.

closed 2019-03-12 by matlo

slmaxk, 2019-03-12:
Hello,

i think it´s your source code? can you tell me where i can add the plugin_init function?
Is there any path i need to add?

matlo, 2019-03-12:
https://github.com/matlo/serialusb/search?q=plugin_init&unscoped_q=plugin_init

matlo, 2019-03-12:
A quick search pointed to: https://github.com/capull0/serialusb/tree/portal