matlo/serialusb #11: [suggestion] replace USB to UART adapter with USB shield

eadmaster, 2021-06-01:
This could make the device acts as a standalone adapter without needing a PC.

reading xinput and DS3 controller inputs should be possible with this lib:
https://github.com/felis/USB_Host_Shield_2.0#xbox-libraries

matlo, 2021-06-01:
GIMX leverages the input stack of the computer. This choice provides almost 100% compatibility with existing and future input devices.

matlo, 2021-06-01:
I am not sure if your suggestion is for serialusb or the GIMX adapter.

closed 2021-06-01 by matlo

matlo, 2021-06-01:
In the first case you still want the USB transfers to be saved on the computer (atmega32u4 chip has a very limited amount of memory).

In the second case, you loose the advantage of using the computer input stack, which provides almost 100% compatibility with existing and future devices.